sekigon-gonnoc/Pico-PIO-USB #1: USB 1:2 Mux

Spegs21, 2022-02-07:
I'm looking to implement a USB mux for a keyboard and mouse. I know these exist in IC form with chips such as the [FSUSB42UMX](https://www.mouser.com/ProductDetail/onsemi-Fairchild/FSUSB42UMX?qs=eJLAyKFqilUl3553fx9DLQ%3D%3D) but I thought it would be neat if I could intercept a certain key combo on the keyboard to toggle the mux. I immediately though of the PIO on the RP2040 knowing it would be fast enough for USB FS. I was really happy to see that this library already existed.

With the current implementation would it be possible to create 2 device ports and a host port? Or could I create two device ports and use the native USB as the host? I know you have examples for multiple host ports but I need multiple devices. It looks like it should be possible, I just wanted to check before I get too deep into this.

Thanks for the help!

sekigon-gonnoc, 2022-02-10:
Do you want to use two device ports at the same time? Or do you want to switch between these ports?
If you reinitialize the PIO-USB-device with different settings, you may be able to switch between the two ports.
Another concern is the GND isolation. Directly connecting the GNDs of different computers may harm the device.

Dalayeth, 2022-02-11:
Also interested in this. I wanted to try making an ogx360 with a Raspberry Pi Pico which would need 4 device connections going to the same Xbox and the built in USB acting in host mode. Is it possible to have multiple device connections and if so what would be the max number? I saw that rx uses 2 SM's. If only LS USB is required would it be possible to get this down to 1 so there could be 4 concurrent device connections for the 4 controller inputs to the Xbox?

closed 2022-02-24 by sekigon-gonnoc