sekigon-gonnoc/Pico-PIO-USB #5: Question: QMK USB-USB converter possible?

raghur, 2022-02-19:
Came to know of this via HN. 

So question around re-purposing this for a usb-usb converter (the hasu converter type which are mostly pro-micro + usb A shield are pretty limiting). Is that a possibility?

1. basically hook up a type A port to the PIO Usb
3. Pi Pico connects via micro usb as usual and reports as a HID device
2. QMK converter fw... reads the PIO USB and reports key strokes to host

I suppose it all depends on what interface the PIO USB exposes to the uC - if it can be like a 'virtual SPI' (no idea if that's even a thing - limited hw/uc knowledge) then the QMK fw could be probably made to work?

Anyway - great project... Amazing work!

sekigon-gonnoc, 2022-02-21:
Yes.
I've made USB-USB converter using rp2040 with usb host IC, and I'm considering use PIO host for the next version.
https://nogikes.booth.pm/items/2256612

closed 2022-02-24 by sekigon-gonnoc

nahuel, 2022-10-04:
Hi @sekigon-gonnoc , is the code/schematics for your USB-USB converter available somewhere?