sekigon-gonnoc/Pico-PIO-USB #8: Dropped input reports

jfedor2, 2022-02-23:
Hi,

While evaluating this project I encountered an issue with HID input reports seemingly dropped. I'm not sure it it's a problem with this library or if I'm doing something wrong.

I have a USB mouse that sends 1000 input reports per second. In the HID report there's a counter field that increases by 1 with every consecutive report. Here's what it looks like when connected directly to the computer (usbhid-dump output). Note the consecutive numbers:

```
002:006:000:STREAM             1645632911.827409
 00 00 00 00 00 81 23

002:006:000:STREAM             1645632911.828371
 00 00 00 00 00 82 23

002:006:000:STREAM             1645632911.829447
 00 00 00 00 00 83 23

002:006:000:STREAM             1645632911.830400
 00 00 00 00 00 84 23

002:006:000:STREAM             1645632911.831391
 00 00 00 00 00 85 23

002:006:000:STREAM             1645632911.832444
 00 00 00 00 00 86 23
```

But when I connect it to my Raspberry Pi Pico and run the "capture_hid_report" example, here's what I get (note that some of the numbers in the counter field are missing - they are not consecutive):

```
cafe:baba EP 0x81:	00 00 00 00 00 3f 0e 
cafe:baba EP 0x81:	00 00 00 00 00 43 0e 
cafe:baba EP 0x81:	00 00 00 00 00 46 0e 
cafe:baba EP 0x81:	00 00 00 00 00 4a 0e 
cafe:baba EP 0x81:	00 00 00 00 00 4e 0e 
cafe:baba EP 0x81:	00 00 00 00 00 51 0e 
cafe:baba EP 0x81:	00 00 00 00 00 55 0e 
cafe:baba EP 0x81:	00 00 00 00 00 58 0e 
```

It seems like I'm only getting every third or fourth input report.

Any help?

sekigon-gonnoc, 2022-02-24:
I have confirmed that the library can send/receive reports at 1ms intervals as shown in the picture.
Printf via USB-CDC may be too slow to show all packets every 1ms.

[image]

jfedor2, 2022-02-24:
Of course you're right!

It was because of the printfs.

Thank you.

closed 2022-02-24 by jfedor2