sekigon-gonnoc/Pico-PIO-USB #13: Writing to flash

jfedor2, 2022-02-26:
I'm making some progress with the library and need a little help again. I'm trying to persist some config information in flash while listening for input reports on a USB port.

The flash writing code looks something like this:

```
    uint32_t ints = save_and_disable_interrupts();
    flash_range_erase(CONFIG_OFFSET_IN_FLASH, FLASH_SECTOR_SIZE);
    flash_range_program(CONFIG_OFFSET_IN_FLASH, buffer, FLASH_PAGE_SIZE);
    restore_interrupts(ints);
```

I noticed that this breaks the code that's receiving data via USB. It's possibly related to the fact that the other core is executing code from flash. I've tried using the multicore locking mechanism described here:

https://raspberrypi.github.io/pico-sdk-doxygen/group__multicore__lockout.html

But it doesn't seem to help.

Any idea on how I could do this?

jfedor2, 2022-02-28:
A workaround that seems good enough for me at the moment is adding this to CMakeLists.txt:

`set(PICO_COPY_TO_RAM 1)`

sekigon-gonnoc, 2022-03-01:
Please try to insert `pio_usb_host_stop()/pio_usb_host_restart()` before and after the flash operation.
This may allow the flash operation without fault, but cause resetting USB connection if flash operation is longer than 1ms.

jfedor2, 2022-03-01:
A disconnect/reconnect wouldn't be an issue, but I can't get it to work. Even doing just:
```
pio_usb_host_stop();
pio_usb_host_restart();
```
with nothing in-between (no writing to flash) seems to break the USB connection permanently.

But given the above workaround it's not currently critical for me.

sekigon-gonnoc, 2022-03-05:
I found that second argument of `alarm_pool_create()` must be set greater than 1 in order to use the stop/resume function.

https://github.com/sekigon-gonnoc/Pico-PIO-USB/blob/a09cf8917ec66cb45910a0e87a7851d80c55b0cf/example/capture_hid_report/capture_hid_report.c#L18