sekigon-gonnoc/Pico-PIO-USB #19: USB Communication problem (newbie question)

mnesarco, 2022-03-29:
**Describe the bug you encountered:**
After clone, build and flash, used minicom to see, at leasts "hello!" should appear right? But there is no communication

```
minicom -D /dev/ttyACM0

Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyACM0, 11:13:37

Press CTRL-A Z for help on special keys

control in[timeout]
Enumeration failed(-2)
Root 0 disconnected
Disconnect                                                                                                                                                                
Disconnect device 0                                                                                                                                                       
Root 0 connected                                                                                                                                                          
Device 0 Connected                                                                                                                                                        
control in[complete]                                                                                                                                                      
Enumerating 10c4:8108, class:0, address:1                                                                                                                                 
control out[complete]                                                                                                                                                     
control in[timeout]                                                                                                                                                       
Failed to get string descriptor (Manufacture)                                                                                                                             
control in[timeout]                                                                                                                                                       
Failed to get string descriptor (Product)                                                                                                                                 
control in[timeout]                                                                                                                                                       
Enumeration failed(-2)                                                                                                                                                    
Root 0 disconnected                                                                                                                                                       
Disconnect                                                                                                                                                                
Disconnect device 1                                                                                                                                                       
Root 0 connected                                                                                                                                                          
Device 0 Connected                                                                                                                                                        
control in[complete]                                                                                                                                                      
Enumerating 10c4:8108, class:0, address:1
control out[complete]
control in[complete]
control in[timeout]
Failed to get string descriptor (Manufacture)
control in[complete]
control in[timeout]
Failed to get string descriptor (Product)
control in[complete]
control in[timeout]
Enumeration failed(-2)
Root 0 disconnected
Disconnect
Disconnect device 1
Root 0 connected
Device 0 Connected
control in[complete]
Enumerating 10c4:8108, class:0, address:1
control out[complete]
control in[complete]
control in[timeout]
Failed to get string descriptor (Manufacture)
control in[timeout]
Failed to get string descriptor (Product)
control in[timeout]
Enumeration failed(-2)
Root 0 disconnected
Disconnect
Disconnect device 1
Root 0 connected
Device 0 Connected
control in[complete]
Enumerating 10c4:8108, class:0, address:1
control out[complete]
control in[timeout]
Failed to get string descriptor (Manufacture)
control in[timeout]
Failed to get string descriptor (Product)
control in[timeout]
Enumeration failed(-2)
Root 0 disconnected
Disconnect
Disconnect device 1
Root 0 connected
Device 0 Connected
```

**What did you expect to happen instead?**

I expected the connection to succeed and to see "hello!" at least.

**What actually happened in your case?**

No communication.

**Additional information**

```
git clone https://github.com/sekigon-gonnoc/Pico-PIO-USB.git
cd Pico-PIO-USB/example
mkdir build
cd bulild
cmake ..
make
```

Then copied: Pico-PIO-USB/example/build/capture_hid_report/capture_hid_report.uf2 to Pico board and Disconnect/Reconnect.

touchgadget, 2022-03-29:
Have you tried VBUS instead of VSYS? https://github.com/sekigon-gonnoc/Pico-PIO-USB/discussions/7

mnesarco, 2022-03-29:
Tried VBUS:
```
Root 0 connected
Device 0 Connected
control in[complete]
Enumerating 10c4:8108, class:0, address:1
control out[complete]
control in[complete]
control in[complete]
Manufacture:YSPRINGTECH
control in[timeout]
Failed to get string descriptor (Product)
control in[timeout]
Enumeration failed(-2)
Root 0 disconnected
```

mnesarco, 2022-03-29:
It worked with a different mouse. Both mouses do work connected directly to the computer, but only one mouse works with the Pico. So basically this issue is a duplicate of #18

mnesarco, 2022-03-31:
I am closing this because this is a duplicate of #18

closed 2022-03-31 by mnesarco