sekigon-gonnoc/Pico-PIO-USB #31: 💡[FEATURE] Make repo Arduino library compatible

hathach, 2022-06-09:
Hi hi, I am trying to get pico-pio-usb working with arduino platform, basically with
- https://github.com/earlephilhower/arduino-pico
- https://github.com/adafruit/Adafruit_TinyUSB_Arduino

It is actually quite easy to make this repo compatible with Arduino IDE. We only need to
- add `library.properties`
- rename example to `examples` (plugrals)
- also included generated `usb_rx/tx.pio.h` since we probably don't have access to cmake `pico_generate_pio_header`
- Finally registry to arduino indexer https://github.com/arduino/library-registry

Another gotcha that Arduino IDE will recursive compiles all the files/folders under src, but I think we are good with it, just need to pay to this attention in the future PRs. But we could add an ci to verify that it build with arduino just fine.

Let me know what you think, since I have done a decent amount of work with Arduino library, I could help making all of these.

@ladyada

sekigon-gonnoc, 2022-06-13:
LGTM

hathach, 2022-06-13:
thanks, I will put things together and make an PR when it is ready.

sfranzyshen, 2022-06-16:
https://github.com/adafruit/Adafruit_TinyUSB_Arduino/releases/tag/1.11.0

1.11.0 - Updated tinyusb to support rp2040 pio usb host

hathach, 2022-06-16:
No, that is part of the auto release schedule of adafruit in weekly basic. There are a few PRs need to merge first, along with adding this library to arduino lib indexer.

Colatino, 2022-06-16:
Really looking forward to see this work along with arduino-pico

hathach, 2022-06-20:
implemented by #35

closed 2022-06-20 by hathach

Jasoroony, 2022-08-18:
I'm really keen to make a Pico based generic USB comms cable for use between 2 computers (I.E. my car's touch screen which I can write software for, and a raspberry pi running Libreelec/kodi - which unfortunately doesn't support the Pi-Zero & Pi-4B gadget mode drivers for a CDC device).  This would present as a COM port 'device' on both machines and data could be written and read as needed for them to communicate.  My thoughts are, to configure my Pico W (which just arrived today) for TinyUSB Serial using the earlephilhower/arduino-pico platform (that's actually working quite well, in the 2 hours I've been playing with it) and add another Serial device using the Pico-PIO-USB support now added there.

It seems I may be a bit early though on this sort of project - so I was wanting to ask for any advice on how I could progress the PIO half of what I have in mind?  I've played a bit with the low-level TinyUSB CDC example (bypassing the buffering) and it would seem like an easy thing for the Pico to just read USB packets and write them "as-is" to the other device - making this very efficient...  But I'm a bit lost at the moment how to progress it all?

Anyway, thanks for the inspiration on this, it's all very interesting.

Jasoroony, 2022-08-29:
Well, I got it to compile with a lot of effort, then link with a bit more effort, then run (not @ 120MHz, but only @ 240MHz) where it falls over fairly quickly once I transmit a 64 byte packet in both directions in sequence (device returns what data it gets, being 64 bytes in this test case).  I might try 8 byte packets later some time, that seems to work (based only on what I see when the config/mount stuff is happening when it's plugged in).