sekigon-gonnoc/Pico-PIO-USB #40: Missing dcd_pio_usb.c

Timu5, 2022-07-02:
When I try to run `cmake ..` I get this:
```
PICO_SDK_PATH is /home/user/pico-sdk
PICO platform is rp2040.
Build type is Release
PICO target board is pico.
Using board configuration from /home/user/pico-sdk/src/boards/include/boards/pico.h
TinyUSB available at /home/user/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
cyw43-driver available at /home/user/pico-sdk/lib/cyw43-driver
lwIP available at /home/user/pico-sdk/lib/lwip
-- Configuring done
CMake Error at host_hid_to_device_cdc/CMakeLists.txt:2 (add_executable):
  Cannot find source file:

    /home/user/pico-sdk/lib/tinyusb/src/portable/raspberrypi/pio_usb/dcd_pio_usb.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

Quick workaround is to disable host_hid_to_device_cdc in CMakeLists.txt like this:
```
add_subdirectory(capture_hid_report)
add_subdirectory(usb_device)
# add_subdirectory(host_hid_to_device_cdc)
```

But where can an individual obtains missing files to get all examples to work?

sfranzyshen, 2022-07-02:
this file is in the tinyusb package ... a lot has been happening amount various packages lately ... you may need to update stuff to the latest greatest ... 

https://github.com/hathach/tinyusb/tree/master/src/portable/raspberrypi/pio_usb

Timu5, 2022-07-02:
Thanks, working now :)

closed 2022-07-02 by Timu5