sekigon-gonnoc/Pico-PIO-USB #53: USB Hub causes a CONTROL_ERROR when plugged in

drflamemontgomery, 2022-11-22:
I noticed that when using a USB Hub with the capture_hid_report example, it  would cause a CONTROL_ERROR. I found that by adding a wait of 1000us in the control_in_protocol function, it would allow the USB Hub to initialize without erroring.
```c
//=======================================
//    src/pio_usb_host.c
//=======================================
static int __no_inline_not_in_flash_func(control_in_protocol)(
    usb_device_t *device, uint8_t *tx_data, uint16_t tx_length,
    uint8_t *rx_buffer, uint16_t request_length) {
  int res = 0;

  control_pipe_t *pipe = &device->control_pipe;

  busy_wait_us(1000); // Adding this line fixes the usb hub
```
I am using USB Female Module and a USB High Speed Hub with 4 Ports

drflamemontgomery, 2023-04-27:
I unfortunately never found the cause of the behavior.
Have you tried using busy_wait_us or busy_wait_ms?

drflamemontgomery, 2023-04-27:
Would you also be able to print a log of the USB using the capture_hid_report program

knflrpn, 2024-06-27:
I was trying to connect to a Nintendo Switch Pro Controller and was running into this same issue, and the busy_wait_us(1000); change fixed it for me.

Interestingly, the controller would _always_ connect on the third attempt (removing and reinserting the USB cable) after resetting the Pico.  But the fix above made it connect on the first attempt.