sekigon-gonnoc/Pico-PIO-USB #55: 🪲[BUG] First Packet in the Data Stage of a Control Transfer is Data0

soulfish5, 2022-11-26:
I was using the PICO as a USB Host for a radio, using the built-in USB connector. I switched to the PIO implementation and the device no longer worked.  

Setup of the radio requires Control Transfers with data stages.

Using a Logic Analyzer I compared the output from the PICO when using the built-in connector and PIO. The data stage of the control transfer starts with Data1 on the working built-in USB connector.
The data stage of the control transfer starts with Data0 when using PIO. 

[https://www.keil.com/pack/doc/mw/USB/html/_u_s_b__control__transfers.html
](url) states:  "The DATA stage is optional. If present, it always starts with a transaction containing a DATA1 packet. Then, the transaction type alternates between DATA0 and DATA1 until all required data have been transferred."

The radio works and the first packet of the data stage is set correctly after I set  ep_data_id to 1 instead of 0 here: [https://github.com/sekigon-gonnoc/Pico-PIO-USB/blob/main/src/pio_usb_host.c#L515](url)

I have a branch I can use to make a merge request if that looks like a good solution. 
Thanks.

soulfish5, 2022-11-26:
Proposed change doesn't make sense.  Will check other code to see if the issue lies outside of this library.

hathach, 2022-12-22:
#58 should fix this issue, @soulfish5 would you mind trying it out ?

soulfish5, 2022-12-25:
@hathach I should be able to give it a try tomorrow. Thanks!

hathach, 2022-12-26:
@soulfish5 no rush, take your time, It is holidays :)

soulfish5, 2023-01-06:
> @soulfish5 no rush, take your time, It is holidays :)

Holidays were thoroughly enjoyed!  Issue tested and resolved. Thanks!

PR #58 (2023-01-07): fix control data & status stage always start with DATA1
control data & status stage always start with DATA1. This pr fixes an issue with an Control OUT transfer (with data) where the PID is incorrectly DATA0. This should fix #55 

PS: This issue is kind of critical for me to get host cdc running (need SET LINE CODING). @sekigon-gonnoc if possible, would you mind making a release after reviewed/merged. Thank you in advance.
closed 2023-01-06 by soulfish5