sekigon-gonnoc/Pico-PIO-USB #61: 💡[FEATURE] MIDI support

marchingband, 2022-12-29:
So few MCUs offer USB MIDI HOST support, it would be AMAZING to have RPI2040 support for that. If possible, could you add an example that shows how to implement this?
Thanks!!!

deepbluemind74, 2022-12-30:
Yes. See [pico-usb-midi-filter](https://github.com/rppicomidi/pico-usb-midi-filter) for an example.

marchingband, 2022-12-30:
Wow also https://github.com/rppicomidi/midi2usbhost which uses the USB peripheral.

marchingband, 2023-05-06:
Is this library able to be "pluggable"? ie. can a device be un-plugged, and another plugged-in gracefully?

KBLiveSolutions, 2023-12-14:
hey there
I'm on my way to building a USB MIDI host with my RP2040 and Arduino.
I managed to get the device_info sketch working, but I'm not sure what would be the next step to get MIDI controllers recognized and working.
The rest of my code is Arduino, so it would be better if I stayed with it.
Can anyone help?