sekigon-gonnoc/Pico-PIO-USB #63: 💡[FEATURE] host_hid_to_device_cdc example - controlling keyboard LEDs

samartzidis, 2022-12-30:
Hi,

In the _host_hid_to_device_cdc_ example, you mention in a comment:
`// TODO control LED on keyboard of host stack`

In the above example, I tried playing with the keyboard Caps Lock LED using the _tuh_hid_set_report_ function like so:

```
static uint8_t leds = KEYBOARD_LED_CAPSLOCK; //  = 2
tuh_hid_set_report(dev_addr, instance, 0, HID_REPORT_TYPE_OUTPUT, &leds, sizeof(leds));
```

but failed to get any results.

Is it because I am not doing things the right way or is it something that the _Pico-PIO-USB_ library does not yet support?

Thank you.

sekigon-gonnoc, 2022-12-30:
This may help you #64

samartzidis, 2022-12-30:
Thank you so much for the quick reply!

The code you've added calls the _tuh_hid_set_report_ on CAPS LOCK key code received and I have verified that it gets called. Nonetheless, it still fails to light the CAPS LOCK led up. If this code worked for you then it definitely has to do with my specific keyboard or possibly with the version of the tinyusb library. I will give it a try with different USB keyboards first.

samartzidis, 2023-07-26:
The error was with my setup. On second attempt I can verify that this now worked. Thanks.

closed 2023-07-26 by samartzidis