sekigon-gonnoc/Pico-PIO-USB #66: 🪲[BUG] Can not communicate with Arduino nano every

Icesythe7, 2023-01-18:
At this point I have been at this for days so here is the issue, I have this code on an arduino nano 
```cpp
void setup() {
  Serial.begin(115200);
}

void loop() {
  while (Serial.available())
  {
    const uint8_t res = Serial.read();
    if (res == 'V')
      Serial.println("4.7V"); 
  }
  Serial.println("test");
  delay(1500);
}
```

and this is the code on the pico
```cpp
// pio-usb is required for rp2040 host
#include "pio_usb.h"

// TinyUSB lib
#include "Adafruit_TinyUSB.h"

// Pin D+ for host, D- = D+ + 1
#define HOST_PIN_DP       2

// USB Host object
Adafruit_USBH_Host USBHost;

// CDC Host object
Adafruit_USBH_CDC  SerialHost;

//--------------------------------------------------------------------+
// Setup and Loop on Core0
//--------------------------------------------------------------------+

void setup() {
  Serial1.begin(115200);

  Serial.begin(115200);
  while ( !Serial ) delay(10);   // wait for native usb

  Serial.println("TinyUSB Host Serial Echo Example");
}

void loop()
{
  //Serial.printf("%d\n", SerialHost.connected());
  uint8_t buf[64];

  // Serial -> SerialHost
  if (Serial.available()) {
    size_t count = Serial.read(buf, sizeof(buf));
    if ( SerialHost && SerialHost.connected() ) {
      Serial.printf("sending %s\n", buf);
      SerialHost.write(buf, count);
      SerialHost.flush();
    }
  }

  // SerialHost -> Serial
  if ( SerialHost.connected() && SerialHost.available() ) {
    size_t count = SerialHost.read(buf, sizeof(buf));
    Serial.printf("%s\n", buf);
  }
}

//--------------------------------------------------------------------+
// Setup and Loop on Core1
//--------------------------------------------------------------------+

void setup1() {
  while ( !Serial ) delay(10);   // wait for native usb
  Serial.println("Core1 setup to run TinyUSB host with pio-usb");

  // Check for CPU frequency, must be multiple of 120Mhz for bit-banging USB
  uint32_t cpu_hz = clock_get_hz(clk_sys);
  if ( cpu_hz != 120000000UL && cpu_hz != 240000000UL ) {
    while ( !Serial ) {
      delay(10);   // wait for native usb
    }
    Serial.printf("Error: CPU Clock = %u, PIO USB require CPU clock must be multiple of 120 Mhz\r\n", cpu_hz);
    Serial.printf("Change your CPU Clock to either 120 or 240 Mhz in Menu->CPU Speed \r\n", cpu_hz);
    while(1) {
      delay(1);
    }
  }

#ifdef HOST_PIN_VBUS_EN
  pinMode(HOST_PIN_VBUS_EN, OUTPUT);

  // power off first
  digitalWrite(HOST_PIN_VBUS_EN, 1-HOST_PIN_VBUS_EN_STATE);
  delay(1);

  // power on
  digitalWrite(HOST_PIN_VBUS_EN, HOST_PIN_VBUS_EN_STATE);
  delay(10);
#endif

  pio_usb_configuration_t pio_cfg = PIO_USB_DEFAULT_CONFIG;
  pio_cfg.pin_dp = HOST_PIN_DP;
  USBHost.configure_pio_usb(1, &pio_cfg);

  // run host stack on controller (rhport) 1
  // Note: For rp2040 pico-pio-usb, calling USBHost.begin() on core1 will have most of the
  // host bit-banging processing works done in core1 to free up core0 for other works
  USBHost.begin(1);
}

void loop1()
{
  USBHost.task();

  // periodically flush SerialHost if connected
  if ( SerialHost && SerialHost.connected() ) {
    SerialHost.flush();
  }
}

void tuh_mount_cb(uint8_t daddr) {
  Serial.printf("Device attached, address = %d\r\n", daddr);

  if (tuh_cdc_mounted(0)) {
    Serial.printf("Device at %d is a cdc device!\n", daddr);
    cdc_line_coding_t line_encoding;
    if (tuh_cdc_get_local_line_coding(0, &line_encoding))
    {
      Serial.printf("baud = %d\n", line_encoding.bit_rate);
      Serial.printf("data bits = %d\n", line_encoding.data_bits);
      Serial.printf("stop bits = %d\n", line_encoding.stop_bits);
      Serial.printf("parity = %d\n", line_encoding.parity);
    }
  } else {
    Serial.printf("Device at %d is NOT cdc DEVICE!\n", daddr);
  }

  // Get Device Descriptor
  //tuh_descriptor_get_device(daddr, &desc_device, 18, print_device_descriptor, 0);
}

//--------------------------------------------------------------------+
// TinyUSB Host callbacks
//--------------------------------------------------------------------+

// Invoked when a device with CDC interface is mounted
// idx is index of cdc interface in the internal pool.
void tuh_cdc_mount_cb(uint8_t idx) {
  // bind SerialHost object to this interface index
  SerialHost.begin(idx);

  Serial.printf("SerialHost is connected to a new CDC device (%d)\n", idx);
}

// Invoked when a device with CDC interface is unmounted
void tuh_cdc_umount_cb(uint8_t idx) {
  if (idx == SerialHost.getIndex()) {
    // unbind SerialHost if this interface is unmounted
    SerialHost.end();
    Serial.println("SerialHost is disconnected");
  }
}
```
I plug in the nano via its usb port to the gpio 2 and 3 of pico and it is recognized as a cdc device and does connect yet I can not communicate with it 
```
SerialHost is connected to a new CDC device (0)
Device attached, address = 1
Device at 1 is a cdc device!
baud = 0
data bits = 0
stop bits = 0
parity = 0
```
I have tried changing this  ```
#define CFG_TUH_CDC_LINE_CODING_ON_ENUM   { 115200, CDC_LINE_CONDING_STOP_BITS_1, CDC_LINE_CODING_PARITY_NONE, 8 }
```
in the config thinking maybe it was a baud issue but then isConnected() just returns false for SerialUSB..furthermore if you try this same thing on an arduino uno it doesnt even show up as a cdc device...can someone post an example of how to simply plug in a device and communicate over serial like u would if you plugged it into the pc?

AshwinDeTaeye, 2023-02-13:
Hello, have you found a solution yet?

Icesythe7, 2023-02-13:
> 

I have not, let me know if you do