sekigon-gonnoc/Pico-PIO-USB #83: [bug?] usb 1.0 Low-Speed device detection in host mode / lsusb.ino

tobozo, 2023-04-30:
hi and thanks for this awesome library :+1: 

I'm the maintainer of [ESP32-USB-Soft-Host](https://github.com/tobozo/ESP32-USB-Soft-Host). 
This library does not use tinyUSB, nor does it use the builtin usb chip that comes with some variants of the esp32 (s2/s3), also it is limited to a few set of pins.
It is similar to Pico-PIO-USB in a sense it's also a wrapper around a third party usb library, only it is limited to usb 1.0 Low-Speed (1.5Mbps).

I was very happy to find that Pico-PIO-USB works out of the box and supports usb 1.1, so I expanded the device_info example sketch to match the precision of linux `lsusb -v` system command, and wrote down the list of successfully detected devices:

- Akai MPK Mini midi keyboard (hidraw/audio stream, midi stream)
- 32GB key (mass storage)
- sdcard adapter (smart card)
- audio dongle (headphone+mic)
- cp21xx/ch34x uart from random devices (pico, esp32, arduino, esp8266)
- yubico key (hid)
- android phone (imaging+mass storage)

**However none of the usb 1.0 Low-speed device I tried got detected.**

This log shows a low cost keyboard from pimoroni as seen from linux, works with ESP32-USB-Soft-Host but can't get detected by Pico-PIO-USB:

```log
april 30 10:58:35 kernel: usb 1-3.4.3: new low-speed USB device number 97 using xhci_hcd
april 30 10:58:36 kernel: usb 1-3.4.3: new low-speed USB device number 98 using xhci_hcd
april 30 10:58:37 kernel: usb 1-3.4.3: New USB device found, idVendor=1c4f, idProduct=0002, bcdDevice= 1.10
april 30 10:58:37 kernel: usb 1-3.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
april 30 10:58:37 kernel: usb 1-3.4.3: Product: USB Keyboard
april 30 10:58:37 kernel: usb 1-3.4.3: Manufacturer: SIGMACHIP
april 30 10:58:37 kernel: input: SIGMACHIP USB Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4.3/1-3.4.3:1.0/0003:1C4F:0002.001D/input/input21
april 30 10:58:37 kernel: hid-generic 0003:1C4F:0002.001D: input,hidraw6: USB HID v1.10 Keyboard [SIGMACHIP USB Keyboard] on usb-0000:00:14.0-3.4.3/input0
april 30 10:58:37 kernel: input: SIGMACHIP USB Keyboard Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4.3/1-3.4.3:1.1/0003:1C4F:0002.001E/input/input22
april 30 10:58:37 mtp-probe[245579]: checking bus 1, device 98: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4.3"
april 30 10:58:37 mtp-probe[245579]: bus: 1, device: 98 was not an MTP device
april 30 10:58:37 kernel: input: SIGMACHIP USB Keyboard System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3.4/1-3.4.3/1-3.4.3:1.1/0003:1C4F:0002.001E/input/input23
april 30 10:58:37 kernel: hid-generic 0003:1C4F:0002.001E: input,hidraw7: USB HID v1.10 Device [SIGMACHIP USB Keyboard] on usb-0000:00:14.0-3.4.3/input1
```

Is this behaviour by design (e.g. tinyUSB can't do usb 1.0 and 1.1 simultaneously), unexpected, or just unimplemented?

sekigon-gonnoc, 2023-05-02:
This library is designed to communicate with USB LS and FS devices, and the speed is automatically detected, so no software changes are required. However, it has been found that communication with some devices (especially LS devices with poor clock accuracy) may not work well (#18). This issue has been improved in version 0.5.2, but there is still a possibility of problems occurring.

tobozo, 2023-05-02:
thanks for your reply :+1: 

> and the speed is automatically detected

Oh that's good news, the parent library had me [doubting about that](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/src/portable/raspberrypi/pio_usb/hcd_pio_usb.c#L91).

> This issue has been improved in version 0.5.2

When running my sketch it looks like there's no usb-ls support at all, so I'll check my own code for bugs first :wink:

shreeve, 2023-07-26:
@tobozo - What did you find out here?

tobozo, 2023-07-27:
@shreeve I got lost while trying to setup probes and debug tools as suggested in #18 and haven't had to opportunity to continue the research since. I won't get hands on my 'lab' and research code until the end of this summer so I'm afraid I have nothing new to share for the meantime.