sekigon-gonnoc/Pico-PIO-USB #99: 🪲[BUG] USB Device not being recognised by host

greateggsgreg, 2023-09-05:
**Describe the bug you encountered:**

The `examples/usb_device` example does not present as a USB device to my MacOS host.

**What did you expect to happen instead?**

I expected it to be enumerated, show up as `cafe:0000` in `lsusb`, and cause a mouse wiggle every 0.5s.

**What actually happened in your case?**

No device is enumerated. With my logic analyser I can see a regular SOF frames every 1 millisecond, and a reset every second - suggesting the circuit is physically valid, and the firmware is running on device.

[image]

https://github.com/sekigon-gonnoc/Pico-PIO-USB/blob/main/src/pio_usb_device.c#L368

The reset suggests to me the Pico does not have an accurate reading of the GPIO pins as it is reading the pair as low, but oddly is able to drive them to reset the connection.

**Additional information**

I have the Pico wired up as:
GPIO 0 -> 1.5 kΩ pull up to 3.3v -> D+
GPIO 1 -> D-
Pico GND -> USB GND

I have tried this with another Pico just to sanity check, as well as confirmed the version of Pico-SDK is contemporary with the examples commit date.

Any guidance you can provide to debug further would be appreciated.

---

asit-fdraschbacher, 2024-10-18:
Unfortunately, I'm observing the same problem: The Pico-PIO-USB USB device is not recognised by macOS hosts. Communication already fails while querying the device descriptor.

Did you find any solution to this issue?

sapfeer0k, 2025-01-07:
I also have the same problem. MacOS reports about wrong descriptor. I have some logs, but I do not have USB analyzer to verify USB packets.