sekigon-gonnoc/Pico-PIO-USB #106: [QUESTION] Is it possible to use Pico-PIO-USB but mux the pins to the USB Phy microUSB connector?

shreeve, 2023-12-19:
The Pico's have a built in USB Phy and a microUSB connector, along with internal logic for handling the USB communications. However, Pico-PIO-USB is so cool, that it would be nice to be able to use it's logic (and the ability to control what it's doing more easily) instead of the USB Phy built in to the Pico.

Based on the Errata 5 workaround for a hardware issue with the Pico's, I can see that they are muxing the DP/DM pins to solve and issue with the hardware. Thus, it looks possible to re-map the DM/DP pins on to the physical connector and bypass the built-in, hardware based, native USB Phy and have it effectively "replaced" with a pure Pico-PIO-USB version.

Is this possible??? If not, I'm stuck having to solder or otherwise connect to different GPIO pins on each board that we prepare and send out to the field. If we could "commandeer" and take over the microUSB connector on the Pico, but drive it with Pico-PIO-USB, then I wouldn't have to do any soldering or otherwise connecting up wires, etc. and it would be awesome. :-)

Thoughts?

shreeve, 2023-12-22:
@hathach or @sekigon-gonnoc - Do you know if this is possible?

hathach, 2024-01-09:
@shreeve No I don't think it is possible to mux different DM/DP pin. The usb_muxing is refering to using different PHY but still using the same pins, much like switching the web back-end from azure to aws (phy), while the front-end (dm/dp pin) is still the same.

If muxing is required, you can use addtional muxing IC e.g https://www.adafruit.com/product/5871

[image]

fl-kmarston, 2024-10-16:
@shreeve I think you might be doing what I want to do - I want to connect to some devices that have USB PL230 or USB CP210x UARTS.  I want a box with a pico and a USB A that I can connect to via Bluetooth or WiFi and be able to use the serial console of the devices - this then means a phone can access configuration menus running on the single board computers that have the USB UARTS

shreeve, 2024-11-30:
@sekigon-gonnoc / @hathach / @kilograham - Is it now possible with the rp2350 to mux GPIO pins (here controlled by Pico-PIO-USB) to the USB Phy connector?

hathach, 2024-12-05:
@shreeve the simple anwser is No. The muxing make it posssible to `use DM/DP as general GPIO`, but the other way is not possible i.e `assign other GPIO to USB PHY`. That is because USB DM/DP is differential signals, they require dedicated analog/filter/logic/gate attached to the pins to get it right (like pio-usb). Most vendor will jusst locked that to a few selected pins where you can switch, for pico dm/dp is the only pins

To sum up you can use DP/DM as GPIO and can possible use pio-usb with it. But you cannot connect other GPIO to the USBPHY.

[image]

shreeve, 2024-12-05:
@hathach - Thank you Ha!

shreeve, 2024-12-05:
@hathach - Just to clarify, then ... when it says "Use the usb DP and DM pins as GPIO pins instead of connecting them to the USB controller", doesn't this mean that the DP and DM pins *inside the USB Phy* can act as GPIO pins? If they can act as GPIO pins, can we not read from and write to them like GPIO pins?

hathach, 2024-12-05:
I think when using DM/DP as GPIO, you can turn it on/off and read its value like othe normal GPIO. But it is disconnected from USBPHY making USBHY not acccessible.

```                 
              USBPPHY
        mux  /
DM/DP <-----
            \
              GPIO
```