sekigon-gonnoc/Pico-PIO-USB #118: 🪲[BUG] SerialPIO and USB Host conflicting

DisasterAreaDesigns, 2024-06-06:
**Describe the bug you encountered:**

USB Host prevents SerialPIO RX from working.

**What did you expect to happen instead?**

SerialPIO should work at the same time as USB Host

**What actually happened in your case?**

USB Host PIO connected to GP6, 7.  SerialPIO connected to GP28, 29.
If SerialPIO starts first, the RX works correctly but Core1 fails to boot and USB Host does not work.
If USB Host starts first, it works correctly but SerialPIO RX does not.

**Additional information**

This sketch shows the issue, adapted from example from this version.  Set #define USBFIRST to start USB first, comment out to start SerialPIO first.

Changing the USB Host configuration settings to the Pico W compatibility settings does not fix this issue.

```
/* Example sketch altered to include SerialPIO on GP28, GP29
 * USBHost is blocking SerialPIO and vice-versa
 */

// pio-usb is required for rp2040 host
#include "pio_usb.h"
#define HOST_PIN_DP   6   // Pin used as D+ for host, D- = D+ + 1

#include "Adafruit_TinyUSB.h"

#define LANGUAGE_ID 0x0409  // English

// USB Host object
Adafruit_USBH_Host USBHost;

// holding device descriptor
tusb_desc_device_t desc_device;

SerialPIO mySerial(28, 29);

//#define USBFIRST

// the setup function runs once when you press reset or power the board
void setup()
{

  Serial.begin(115200);

#ifndef USBFIRST
  mySerial.begin(31250); // if this is started first, it works but USB fails
#endif

  while ( !Serial ) delay(10);   // wait for native usb

#ifndef USBFIRST
  Serial.println("SERIAL FIRST");
#endif

  Serial.println("TinyUSB Dual Device Info Example");

#ifdef USBFIRST
  Serial.println("USB FIRST");
  delay(3000);
  mySerial.begin(31250); // if this is started last, USB works and Serial fails
#endif

  Serial.println("CORE0 BOOT COMPLETE");

}

void loop()
{

  // test routine for separate RX
  if (mySerial.available() ) {
    // read the incoming byte:
    char incomingByte = mySerial.read();

    // say what you got:
    Serial.print("I received: ");
    Serial.println(incomingByte, DEC);
  }

}

// core1's setup
void setup1() {
  while ( !Serial ) delay(10);   // wait for native usb
  Serial.println("Core1 setup to run TinyUSB host with pio-usb");

  // Check for CPU frequency, must be multiple of 120Mhz for bit-banging USB
  uint32_t cpu_hz = clock_get_hz(clk_sys);
  if ( cpu_hz != 120000000UL && cpu_hz != 240000000UL ) {
    while ( !Serial ) delay(10);   // wait for native usb
    Serial.printf("Error: CPU Clock = %u, PIO USB require CPU clock must be multiple of 120 Mhz\r\n", cpu_hz);
    Serial.printf("Change your CPU Clock to either 120 or 240 Mhz in Menu->CPU Speed \r\n", cpu_hz);
    while (1) delay(1);
  }

  pio_usb_configuration_t pio_cfg = PIO_USB_DEFAULT_CONFIG;
  pio_cfg.pin_dp = HOST_PIN_DP;

// these settings make no difference to the result
#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
  /* https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/46 */
  pio_cfg.sm_tx      = 3;
  pio_cfg.sm_rx      = 2;
  pio_cfg.sm_eop     = 3;
  pio_cfg.pio_rx_num = 0;
  pio_cfg.pio_tx_num = 1;
  pio_cfg.tx_ch      = 9;
#endif /* ARDUINO_RASPBERRY_PI_PICO_W */

  USBHost.configure_pio_usb(1, &pio_cfg);

  // run host stack on controller (rhport) 1
  // Note: For rp2040 pico-pio-usb, calling USBHost.begin() on core1 will have most of the
  // host bit-banging processing works done in core1 to free up core0 for other works
  USBHost.begin(1);

  Serial.println("CORE 1 BOOT COMPLETE");
}

// core1's loop
void loop1()
{
  USBHost.task();
}

//--------------------------------------------------------------------+
// TinyUSB Host callbacks
//--------------------------------------------------------------------+

// Invoked when device is mounted (configured)
void tuh_mount_cb (uint8_t daddr)
{
  Serial.printf("Device attached, address = %d\r\n", daddr);
}

/// Invoked when device is unmounted (bus reset/unplugged)
void tuh_umount_cb(uint8_t daddr)
{
  Serial.printf("Device removed, address = %d\r\n", daddr);
}

```
---

sekigon-gonnoc, 2024-06-06:
PIO USB TX requires 22 instructions and Serial PIO requires 13 instructions. Therefore, they do not work together in this version.

closed 2024-06-06 by sekigon-gonnoc

sekigon-gonnoc, 2024-06-06:
I'm currently developing a version that requires fewer instructions.
https://github.com/sekigon-gonnoc/Pico-PIO-USB/tree/single-pio

DisasterAreaDesigns, 2024-06-06:
Thanks for the info, I have downloaded the single-PIO fork and it is working with my hardware.