sekigon-gonnoc/Pico-PIO-USB #122: PIO USB v0.6.0 no longer enumerates devices

Kokovec, 2024-06-17:
After upgrading the PIO USB library to version 0.6.0 the callback for a HID device mounting (`void tuh_hid_mount_cb`) no longer works.

Kokovec, 2024-06-17:
I should not that this is happening while using the Adafruit Tiny USB library.

sekigon-gonnoc, 2024-06-19:
set the optimize option to something other than `-Os`

Kokovec, 2024-06-19:
This works but breaks other libraries (like NeoPixel).

edwardsharp, 2024-07-02:
> set the optimize option to something other than `-Os`

👋 can you explain how to do that?

fwiw: i stumbled upon this with similar experience, `0.6.0` doesn't run the callback, `0.5.3` seems to still work as i expect.

Kokovec, 2024-07-02:
> > set the optimize option to something other than `-Os`
> 
> 👋 can you explain how to do that?
> 
> fwiw: i stumbled upon this with similar experience, `0.6.0` doesn't run the callback, `0.5.3` seems to still work as i expect.

In the Arduino IDE it's under "Tools->Optimize"

lyusupov, 2024-07-18:
> set the optimize option to something other than `-Os`

@sekigon-gonnoc (with CC: to @hathach ) , is it possible to apply a local area "no-optimize" effect by using `#pragma GCC optimize` https://gcc.gnu.org/onlinedocs/gcc/Function-Specific-Option-Pragmas.html with the  PIO USB library and/or Adafruit Tiny USB library while keeping the rest of source codes (a sketch, Arduino Core, other libraries) compiled with `-Os` flag ?

sweetlilmre, 2024-07-27:
Compiling and running capture\_hid\_report on `v0.6.0` under VSCode in:

*   debug `-Og`
*   release `-O3`

results in the same behaviour: no devices enumerated. `v0.5.3` works as expected. Am I missing something?

(I am not familiar with cmake so I may well be).

sweetlilmre, 2024-07-28:
More info.  I tried a fully unoptimized build `-O0` and a debug build `-Og`. 

I seem to get as far as the device being recognised as connected, but all endpoints are NULL:

```
        if (!device->connected) {
          continue;
        }

        // Print received packet to EPs
        for (int ep_idx = 0; ep_idx < PIO_USB_DEV_EP_CNT; ep_idx++) {
          endpoint_t *ep = pio_usb_get_endpoint(device, ep_idx);

          if (ep == NULL) {
            break;
          }
```

i.e. `ep` is always NULL. This would suggest something else apart from a `-Os` issue is happening.

The device connected to the pico is a PS3 dual shock 3 controller. Standard pin connections (GP0 and GP1), as before the device enumerates (well has valid endpoints) with `v0.5.3`

closed 2024-08-26 by sekigon-gonnoc

kripton, 2024-09-04:
I am experiencing something similar. I tried running the `capture_hid_report` on an "Adafruit RP2040 Feather with USB host" (so I had to adapt the pins to 16 and 17 and also add code to drive GPIO 18 high since that enables power on the DFP. With commit 0f747aaa0c16f750bdfa2ba37ec25d6c8e1bc117 the example works and the connected device (some USB RAW HID device that can be bought online) enumerates fine:
```
Root 0 connected
Device 0 Connected
control in[complete]
Enumerating 0b00:1000, class:0, address:1
...
```

However, with latest main (fe9133fc513b82cc3dc62c67cb51f2339cf29ef7) and exactly the same changes done by me, the enumeration fails:
```
Root 0 connected
Device 0 Connected
control in[error]
Enumeration failed(-1)
```

sweetlilmre, 2024-09-06:
Same results here:
v0.5.3

```
hello!
Root 0 connected
Device 0 Connected
control in[complete]
Enumerating 054c:0268, class:0, address:1
control out[complete]
control in[error]
Failed to get string descriptor (Manufacture)
control in[error]
Failed to get string descriptor (Product)
control in[complete]
control in[complete]
control out[complete]
inum:0, altsetting:0, numep:2, iclass:3, isubclass:0, iprotcol:0, iface:0
        bcdHID:1.11, country:0, desc num:1, desc_type:34, desc_size:148
control out[error]
control in[complete]
                Report descriptor:05 01 09 04 a1 01 a1 02 85 01 75 08 95 01 15 00 26 ff 00 81 03 75 01 95 13 15 00 25 01 35 00 45 01 05 09 19 01 29 13 81 02 75 01 95 0d 06 00 ff 81 03 15 00 26 ff 00 05 01 09 01 a1 00 75 08 95 04 35 00 46 ff 00 09 30 09 31 09 32 09 35 81 02 c0 05 01 75 08 95 27 09 01 81 02 75 08 95 30 09 01 91 02 75 08 95 30 09 01 b1 02 c0 a1 02 85 02 75 08 95 30 09 01 b1 02 c0 a1 02 85 ee 75 08 95 30 09 01 b1 02 c0 a1 02 85 ef 75 08 95 30 09 01 b1 02 c0 c0 
                        epaddr:0x02, attr:3, size:64, interval:1
                        epaddr:0x81, attr:3, size:64, interval:1
```

v0.6.1 / master

```
hello!
Root 0 connected
Device 0 Connected
control in[error]
Enumeration failed(-1)
Root 0 disconnected
Disconnect
Disconnect device 0
```

@sekigon-gonnoc  please can you reopen this issue?

bitcores, 2024-09-23:
I have just started playing with this library, so I can't speak for how it was in v0.5.3, but I am experiencing much the same as the recent commenters.
One thing I have noticed, though, is that if I wait a while after resetting the Pico, or after unplugging the USB device, (about 15-30 seconds) and then plug the USB device in, it tends to enumerate it, but if I plug it in immediately, or unplug and plug quickly, it tends to fail to enumerate.

```control in[complete]
                Report descriptor:06 00 ff 09 01 a1 01 85 10 75 08 95 06 15 00 26 ff 00 09 01 81 00 09 01 91 00 c0 06 00 ff 09 02 a1 01 85 11 75 08 95 13 15 00 26 ff 00 09 02 81 00 09 02 91 00 c0 06 00 ff 09 04 a1 01 85 20 75 08 95 0e 15 00 26 ff 00 09 41 81 00 09 41 91 00 85 21 95 1f 15 00 26 ff 00 09 42 81 00 09 42 91 00 c0
                        epaddr:0x83, attr:3, size:32, interval:2
Root 0 disconnected
Disconnect
Disconnect device 1
Root 0 connected
Device 0 Connected
control in[error]
Enumeration failed(-1)
Root 0 disconnected
Disconnect
Disconnect device 0
Root 0 disconnected
Root 0 connected
Device 0 Connected
control in[error]
Enumeration failed(-1)
Root 0 disconnected
Disconnect
Disconnect device 0
Root 0 disconnected
Root 0 connected
Device 0 Connected
control in[complete]
Enumerating 046d:c52b, class:0, address:1

bitcores, 2024-09-23:
@shuffle2 I tried adding it in a few different places but it didn't seem to help at all.

sapfeer0k, 2024-11-01:
I have the same problem . Using any 0.6.* version USB enumeration fails. 
I am on Windows + VScode Pico extension

bitcores, 2024-11-01:
This particular example (capture hid report) has been deprecated. It's not really supposed to be functional anymore.
I suggest using the host_hid_to_device_cdc example for testing and then as a scaffold for whatever you are building.

PR #132 (2024-08-26): Fix pre-encoded data buffer overflow
Fix #122