sekigon-gonnoc/Pico-PIO-USB #127: 🪲[BUG] Example usb_device stopped working at release 0.6.0

rppicomidi, 2024-08-08:
**Describe the bug you encountered:**
I tested the usb_device example with version 0.6.0. The PIO USB device does not enumerate. I have not attached a USB sniffer to the figure out what is going on, but I suspect it would show no response from the device. Version 0.5.3 worked just fine.
...

**What did you expect to happen instead?**
I expected the PIO USB device to enumerate as a HID device and periodically move the mouse pointer to the right.
...

**What actually happened in your case?**
Nothing that I was able to observe.

...

**Additional information**
I reverted to version 0.5.3 and rebuilt the examples. The PIO USB device worked as expected. I suspect something broke the device interface between 0.5.3 and 0.6.0.

I am using an Ubuntu Linux 22.04LTS host. My pico-sdk is on the develop branch, last updated June 13. TinyUSB is in sync with the master branch. Compiler is `arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)`. Target hardware is a Raspberry Pi Pico board wired to a USB C breakout board. I have set config.pin_dp to 16 before calling `pio_usb_device_init()` because I have DP wired to Pico board pin 21 and DM wired to Pico board pin 22. USB Ground is wired to Pico board pin 23. There is a 1.5k pull-up between Pico board pin 21 and pin 36. I connect the USB C breakout to the test PC via a USB 2.0 hub.

---

lyusupov, 2024-08-13:
duplicates https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/122

rppicomidi, 2024-08-14:
Are you sure this is a duplicate of #122? #122 is about USB Host function. I have no issues with the USB Host interface unless I use optimization -Os. Even debug builds work fine for me. It is the USB Device interface that I am reporting is not working in 0.6.0. For that, no optimization I have tried will suffice.

eriklundh, 2024-09-09:
This is NOT a duplicate of #122!

Jose-Castan, 2024-10-22:
This error still happens at release 0.6.1. 
When connected to the PC, it sometimes mount the mouse or keyboard correctly, but most of the times the error shown using dmesg is 
```console 
usbhid 1-1:1.0: can't add hid device: -71
usbhid: probe of 1-1:1.0 failed with error -71
usbhid 1-1:1.1: can't add hid device: -71
usbhid: probe of 1-1:1.1 failed with error -71
```

Using Beagle Total Phase USB 480 sniffer, the difference when mounting the device one of the times it mounts ok and when error is the following
Mount ok
[image]
Mount error
[image]
When doing the Get Report Descriptor, the error one only sends the first IN txn, but dont sends "C0" one

sapfeer0k, 2024-12-17:
I can confirm, having the same problem :/

braxlan, 2025-03-10:
Is it just the example that is not working or is USB device support generally broken since 0.6.0? I am looking for a working example to find out if it could be a solution for my project.