sekigon-gonnoc/Pico-PIO-USB #135: 🪲[BUG] capture_hid_report does not enumerate Apple Magic Keyboard

peterburk, 2024-08-28:
**Describe the bug you encountered:**
I tried connecting the Apple Magic Keyboard to the HID Remapper project, and it didn't respond. Therefore I cloned the capture_hid_report example, to monitor the serial debug output. I got error CONTROL_ERROR even after adding busy_wait_us(1000);. 

**What did you expect to happen instead?**

I expected the keyboard to enumerate. 

**What actually happened in your case?**

```
hello!Root 0 Connected
Device 0 Connected
control in [error]
Enumeration failed(-1)
Root 0 disconnected
Disconnect
Disconnect device 0

```

**Additional information**

The keyboard does not have a built-in USB hub. I believe that the keyboard chip is the [STM32F103VB](https://www.ifixit.com/Teardown/Magic+Keyboard+Teardown/50995?srsltid=AfmBOoovPeXIm28PWMoG7e4hWay3rlajDaOJv66Di7O2xoCO4IaOQG62). 
---

peterburk, 2024-08-29:
This is the output of a USB analyser when I connect the keyboard directly to the PC:

[image]

This is the output when I connect it to the Pico-PIO-USB:

[image]

peterburk, 2024-08-29:
Changing Pico-PIO-USB\src\usb_definitions.h:309 to

```
#define GET_DEVICE_DESCRIPTOR_REQ_DEFAULT                                      \
  { USB_REQ_DIR_IN, 0x06, 0, 0x01, 0, 0, 0x40, 0 }
```

changed the packet, but it still appears to be malformed. 

[image]

kripton, 2024-09-04:
Can you try an older version of the library? See https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/122#issuecomment-2328632491

peterburk, 2024-09-04:
Thanks for telling me to try the older library! That fixed it, it now enumerates and captures keystrokes as expected :)

closed 2024-09-04 by peterburk

bitcores, 2024-09-23:
Looking at these USB dumps may give a hint at the actual problem that happening here.

In the working capture there is
```
Reset
Power ON
Suspended
Reset
High Speed
GetDescriptor
```
but in the non-working captures there is 
```
Reset
Reset
High Speed
Suspended
GetDescriptor
```
In the pio_usb_host.c we find
```
root->suspended = true; // need a bus reset before operating
```
suggesting that after you suspend the host, you need to reset the bus.
So if this order of events is consistent with the failures to enumerate, it seems that the root is getting reset before it is suspended rather than afterward, perhaps resulting in the root still being suspended when it tries to enumerate the device.