sekigon-gonnoc/Pico-PIO-USB #141: 💡[FEATURE] Document RP2350 support

holly-hacker, 2024-09-15:
**Describe the feature you want to implement here:**
Document whether RP2350 is supported, any any workarounds that may be needed.

---

I'm trying to get this library to work on a Pimoroni Tiny2350 but am having trouble getting devices detected in the `capture_hid_report` example. I noticed that by adding log lines (to get better insight into where the code gets to), it seems to get further which implies that perhaps the Cortex M33 cores in the RP2350 run too fast which messes up some timing.

Could you clarify what the current status on RP2350 support is, even if it is as-of-yet untested?

qsantos, 2026-01-02:
Regarding @PJ5168 report for Waveshare’s RP2350-USB-A, the issue is with [a pull-up resistor (R13)](https://qsantos.fr/2025/11/21/fixing-the-rp2350-usb-a-not-working-as-usb-host/) . Also, [there are no pull-down resistors, which cause other issues](https://qsantos.fr/2026/01/01/the-rp2350-usb-a-cannot-see-devices-disconnect/).

qsantos, 2026-01-07:
Indeed 😢 . I found the [USB interposer](https://www.tindie.com/products/8086net/usb-interposer/), but it looks more barebones. I put my thoughts in [another article](https://qsantos.fr/2026/01/07/no-alternatives-to-the-rp2350-usb-a/).