sekigon-gonnoc/Pico-PIO-USB #145: HOW TO COMPILE THE EXAPLES USING PICO-VISUAL STUDIO CODE?

1250364116, 2024-10-08:
Hello. I am a newbie and want to build host_hdi_to_device_cdc example. what steps should I take to build the example in pico-visual studio?

sapfeer0k, 2024-10-31:
Just open __only__ `examples` folder in the visual studio and then use the extension to compile all the examples.

UsernamePlankalkul, 2024-12-22:
> Just open **only** `examples` folder in the visual studio and then use the extension to compile all the examples.
> 
> Some screenshots: open `examples` directory ![image](https://private-user-images.githubusercontent.com/958172/382009384-c5552821-fe45-45c7-a744-5506dd5dafc9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ4ODY0NDAsIm5iZiI6MTczNDg4NjE0MCwicGF0aCI6Ii85NTgxNzIvMzgyMDA5Mzg0LWM1NTUyODIxLWZlNDUtNDVjNy1hNzQ0LTU1MDZkZDVkYWZjOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMlQxNjQ5MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNjI3ZmJkMWNmNTVhMzA4MDcxNjYyYWFiNzkxZWE1NzY4ZjdhNDQyOGM3MjFlNjE4ZDlmYzE5NjJkMjY5ZmY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.mjLEjuviKftFrfZiSzWkwU2P-VEomU-sPni7ssrIsSg) select pico extension and press compile: ![image](https://private-user-images.githubusercontent.com/958172/382009561-53ac2805-8030-47e3-a10d-0a7c46eff524.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ4ODY0NDAsIm5iZiI6MTczNDg4NjE0MCwicGF0aCI6Ii85NTgxNzIvMzgyMDA5NTYxLTUzYWMyODA1LTgwMzAtNDdlMy1hMTBkLTBhN2M0NmVmZjUyNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMlQxNjQ5MDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMGMwMTk5Njg5NDIxODhlYTBlOGEyYTJhNTE2ZDI0ZDJiMjQ4ZmQ1ZTUzZDZkODNlYzBjNWE5NjM5MDQxMTkyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.tnv8zuyvrpmUd5yT905UZZ75zFe2LFoYhe3lBBT5SR4)

Thanks ... Do you you know which pins should be connected to D+ and D- on Pico RP2040 ?
is it possible to make a short video about this project on youtube ? there are some videos but they dont work actualy ... I used to work with Arduino IDE ... it was very easy but these compiling methods with pico is a bit complicated

PJ5168, 2025-03-14:
Hi,
Failed to build the arduino example under arduion for RP2350 USB A from Waveshare.
Many re-definitions.
Would you please advise what's the right way to build the example using arduino? Thanks!