sekigon-gonnoc/Pico-PIO-USB #149: Hubs no longer work

jfedor2, 2024-11-06:
I noticed that USB hubs stopped working some time recently (when plugged into a Pico acting as a host).

I have bisected the issue down to these two commits: d735828e6964eb0659bf7c727db8d0cc0b30ab62, 36d5be43c7203c75e01cd2397dfa08ff21f6a81c.

rppicomidi, 2025-02-03:
@jfedor2 At least with commit [d17f8ef](https://github.com/sekigon-gonnoc/Pico-PIO-USB/commit/d17f8efe9e6ee266d0b818634ca0ac256825579f), a single hub works fine most of the time. With some drivers, [usb_midi_host](https://github.com/rppicomidi/usb_midi_host) for example, unplug  detection can fail almost all of the time. The polling of hub status immediately after a failed IN or OUT transfer on the unplugged device also fails, and the TinyUSB hub driver never tries again. See issue https://github.com/hathach/tinyusb/issues/2971. As this issue does not happen using the native RP2040 hardware, I would say the issue is most likely in this driver.

rppicomidi, 2025-02-09:
If I roll back the library version to 0.5.3,  https://github.com/hathach/tinyusb/issues/2971 does not happen anymore. I agree with @jfedor2. Something went wrong after release 0.5.3. I tested main (442af43), release 0.6.1, and 0.6.0 and see the same issue.  The last commit that correctly handled a disconnect from a hub while there was a pending transfer was [0f747aa](https://github.com/sekigon-gonnoc/Pico-PIO-USB/commit/0f747aaa0c16f750bdfa2ba37ec25d6c8e1bc117). I know this happens all of the time with a USB MIDI host, and I suspect it happens with USB CDC ACM host, too. @sekigon-gonnoc, any advice on where to look? The change from release 0.5.3 to 0.6.0 was pretty major.

rppicomidi, 2025-05-15:
@hathach @sekigon-gonnoc I am still seeing failed detection of unplug of a MIDI device connected through a USB Hub. My application uses the native TinyUSB MIDI Host driver. I am running  Pico-PIO-USB commit 810653f and TinyUSB commit 542e5b4. Unplugging the hub itself works fine. I am running on a Raspberry Pi Pico W. TinyUSB host runs on core 1. All other code runs on core 0.

hathach, 2025-05-21:
@rppicomidi use latest of both pio-usb and tinyusb and try again.

rppicomidi, 2025-05-21:
@hathach I tried again. I am using Pico PIO USB tag 0.7.2. I am using TinyUSB commit d3a9fee5cbd2490. I see the same result. I can plug in a USB MIDI device directly. Unplug is recognized. I can plug in a MIDI device through a hub. It enumerates and is recognized. If I unplug it, unplug is not recognized. If I unplug the hub, unplug is recognized, but no further plug or unplug events are recognized. My hub enumerates as a Terminus Technology Inc. Hub with vid:pid 1a40:0101.

hathach, 2025-05-22:
can you try to test with
1. plug MIDI device to another hub
2. plug another device to same hub

rppicomidi, 2025-05-22:
I tried entirely new hardware (a new Adafruit Feather RP2040 with USB A Host board), new cables and a new hub. Same result. MIDI device plug in detection is not reliable. MIDI unplug detection never works. I tried another MIDI device. Same result. I repeated with old hub. Same result. MIDI devices were Korg NanoKONTROL2 and Yamaha Reface CS. 

I tried a mass storage device (PNY USB 2.0 32GB flash drive). Plug and unplug detection do not work through a hub. Plug in detection rarely works.

USB Host is running on CPU 1. USB Device is running on CPU 0 (MIDI and CDC UART). Application code is [midi2piousbhub](https://github.com/rppicomidi/midi2piousbhub). Do you have have an example of application code that works reliably with MIDI and USB host on PIO?

rppicomidi, 2025-05-22:
I have also tried again with a much simpler application: [midi2usbhost](https://github.com/rppicomidi/midi2usbhost). I could not test MSC, but I tested all hubs and MIDI devices on an Adafruit Feather RP2040 with USB A Host board.

hathach, 2025-05-23:
can you try a even simpler example like host/device_info in the tinyusb stock example ? using tinyusb tip of master

rppicomidi, 2025-05-23:
OK. I tried host/device_info example. I had to edit the tusb_config.h file to uncomment `#define CFG_TUH_RPI_PIO_USB   1 `
I built the code with
```
make FAMILY=rp2040 BOARD=adafruit_feather_rp2040_usb_host all
```
I programmed the Adafruit Feather RP2040 with USB A Host board with the device_info.uf2 file.

I tried with a USB flash drive. No hub. If I start with nothing plugged in and then plug in, it doesn't always display the device info. But it does display `Device removed, address = 1` when I unplug it. Sometimes it does display the device info. Sometimes, if I reset the board without unplugging, it will display the device info after reset. Same behavior with MIDI.

I do all my testing with powered hubs.
With a hub, sometimes it will display the device info on plug in. Sometimes, I have to reset multiple times. occasionally, it works as expected.

I have tried two different hubs, two different MIDI devices, and a flash drive. Same results: unreliable operation.

If I build the same code to target the RP2040's native USB hardware, it all works as it should with the same  hubs, USB flash drive, and MIDI devices.

Is there log output that would help debug this issue? Do my results differ from yours?

closed 2026-10-23 by sekigon-gonnoc