sekigon-gonnoc/Pico-PIO-USB #151: How to support two USB-HOST ports?

qq8322302, 2024-11-13:
"RP2040, I want to use 4 GPIO pins to simulate two USB-HOSTs, one for connecting a keyboard and one for connecting a mouse."
"Does this project support this? I couldn't find any relevant methods in the code."
"I used a USB-HUB for expansion, but I found that support for some USB-HUB chips is not good. The SL2.1A cannot recognize any connected devices, and the HS8836A cannot properly disconnect and reconnect the second device."

No0ne, 2024-11-27:
You can use this function for that:
```c
int pio_usb_host_add_port(uint8_t pin_dp, PIO_USB_PINOUT pinout);
```

qq8322302, 2024-12-01:
@No0ne  Thank you very much for your reminder. After testing, it has been found that using pio_usb_ host _ ad_port can support multiple host ports.

closed 2024-12-01 by qq8322302

No0ne, 2025-09-04:
I was asking that here: https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/114
I don't think its possible at the momtent.

qq8322302, 2025-09-08:
> > [@No0ne](https://github.com/No0ne) Thank you very much for your reminder. After testing, it has been found that using pio_usb_ host _ ad_port can support multiple host ports.
> 
> Hi Sir, I used this code, but seems like it will stuck on the second host usb when I plug them together ?

Here, I'm conducting the test using the keyboard and mouse. Both of the two ports are functioning properly.

No0ne, 2025-09-08:
Yes two **host** ports work fine but in https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/114 I was asking about two **device** ports..

qq8322302, 2025-09-08:
Multiple device ports. Currently, it seems that they are not supported. We are looking forward to their support in the future.

kbladewht, 2025-09-10:
> Here, I'm conducting the test using the keyboard and mouse. Both of the two ports are functioning properly.

thanks your confirm, it is weird, not sure how to orgnize codes other this one line