sekigon-gonnoc/Pico-PIO-USB #160: 🪲[BUG] rp2350 host mode issue with low speed device

hathach, 2025-02-20:
**Describe the bug you encountered:**

While testing pio-usb host mode with rp2350 using my TinyUSB, I find out that low speed device such as [raspberrypi official mouse](https://www.raspberrypi.com/products/raspberry-pi-mouse/) does not work at all. while it works fine with rp2040. Note: rp2350 does work well with fullspeed device though.

I am testing with pico2 runnning pico-sdk 2.1.1 and latest pico-pio-usb and latest tinyusb   

**What actually happened in your case?**

With raspberry pi mouse, there is just SE0 and SE1 captured on the bus

[image]

While using another generic lowspeed keyboard, I can see an clearer picture. While The IN packet seems to be doing ok, there always is a **300ns SE0 occurs after every OUT and SETUP packet**. (RPI mouse probably panic and corrupt the bus). 

[image]

My best guess is we probably miss something with rp2350 timing. Let me know if you need any extra information on my test or setup

@tannewt

tannewt, 2025-02-25:
#164 fixes this. It was caused by execing an instruction without sideset set. (Therefore 0)

hathach, 2025-04-22:
already fixed by #164

closed 2025-04-22 by hathach