sekigon-gonnoc/Pico-PIO-USB #177: 🪲[BUG] Give a suitable title

thement, 2025-04-14:
**Describe the bug you encountered:**

I believe there's a regression: Raspberry PI Pico (RP2040) running `host_hid_to_device_cdc` example doesn't enumerate Genius USB keyboard **anymore** when upgraded to 0.7.0.
It worked well on ` 3fae64475259bf5f20544413b79ed3641b9df7d9`

**What did you expect to happen instead?**

I expect the example to be fully running, enumerate the keyboard and print key-press.

**What actually happened in your case?**

No events at all.

...

**Additional information**

I can help you debug the issue and bisect the changes if you are interested.

---

sekigon-gonnoc, 2025-04-14:
Could you try 0.7.1?

thement, 2025-04-14:
I tried it just now and it works just fine.

Thank you.

(If you need any more tests just let me know)

closed 2025-04-15 by sekigon-gonnoc